scipp/scippneutron #62: Gravity correction

SimonHeybrock, 2021-04-26:
In scipp/ess/pull/15 @arm61 added [code for gravity corrections](https://github.com/scipp/ess/blob/aa03775d3cb33ead7efb17015c094ccece28ed63/src/ess/reflectometry/corrections.py#L11-L45). Gravity corrections are generic and also used by other techniques, in particular SANS. Therefore it is a good condidate for inclusion in `scippneutron`.

Requirements:
- Computed corrected angle must be supported by subsequent calls to `convert`. This can be achieved, e.g., by storing it as `two_theta`.
- Must support event mode. This would most likely require a change to `convert`, which currently uses only `tof` coords of events, but takes all other params (including `two_theta`) from bin-coords.
- Gravity should be defined as a vector.
- Must work with input geometry defined as in https://scipp.github.io/scippneutron/user-guide/unit-conversions.html#Beamline-geometry-parameters-used-in-unit-conversions. For example, `scattered_beam` instead of `sample_position` and `position`?

Questions:
- Is gravity correction for the primary flight-path required? If so, how?
- Is a generic correction possible, or are there specifics to reflectometry, such as handing of finity beam/sample size?
- In SANS beam is often perpendicular to `\vec g`, but not so in reflectometry? Does it even make sense to have a common implementation?

arm61, 2021-04-26:
Two comments for this: 
1. Do you want me to raise this at the IDS meeting today?
2. If the IDSs think this is an important/worthwhile feature, I can probably work on putting together a document looking at the effect of gravity in the different scattering geometries. Would that be useful to inform implementation?

SimonHeybrock, 2021-04-26:
I think 2) could be valuable, so we can fully understand which aspects need to be corrected. Having a unified mechanism would be a big advantage imho.

arm61, 2021-04-26:
I will add it to the agenda for the meeting today and speak to the others about the requirements.

arm61, 2021-04-26:
It seems there is interest from at least myself and Wojtek for this to be general. I will try and put a document together looking at this this week.

SimonHeybrock, 2021-04-27:
@arm61 Could you write that document as a Jupyter notebook? This will makes things very easy if we get around to implement it, since we can just reuse (parts of) the notebook in the documentation.

arm61, 2021-04-28:
Github won't let me upload a notebook, see Slack.

SimonHeybrock, 2021-04-29:
Thanks, I have placed it in a branch: https://github.com/scipp/scippneutron/blob/gravity-discussion/docs/user-guide/gravity.ipynb

arm61, 2021-04-29:
Cheers, my feeling is that a general detector correction which rebins the data to remove the effect of gravity should be possible. I could put this together in Python but doubt my C++ is up to the level necessary to do it properly.

SimonHeybrock, 2021-04-29:
Maybe I missed it in the notebook, but do we ever need to correct the (angle of the) incident beam for gravity? Or do slits and other components generally ensure that this is irrelevant?

arm61, 2021-04-29:
I don't have this in the notebook yet as it is something I am not 100 % sure of myself. For the Amor code, the assumption is that everything on the detector is specular intensity and therefore $\theta_i = \theta_f$. However, there are some papers from the ILL (see Figure 7 in the attached) which suggest that gravity before the neutron is incident should be accounted for. 
[figaro-focus-point.pdf](https://github.com/scipp/scippneutron/files/6396796/figaro-focus-point.pdf)

arm61, 2021-04-29:
I have just spoken to @ajj about this for SANS and it sounds like there is more complexity. He suggested looking at the Mantid procedures for this.

wpotrzebowski, 2021-06-29:
Relevant Mantid files
https://github.com/mantidproject/mantid/blob/master/Framework/Algorithms/src/GravitySANSHelper.cpp
https://github.com/mantidproject/mantid/blob/7d925b9547bc98235c739200a876e23509eaf492/Framework/Algorithms/src/Q1DWeighted.cpp

OwenArnold, 2021-09-17:
@arm61 @wpotrzebowski  where are we with this? It seems like details are still being worked out regarding how this should work for SANS? Is there anything I can do to expedite this?

SimonHeybrock, 2024-06-24:
@jl-wynen Can we consider this fixed based on your recent additions?

jl-wynen, 2024-06-24:
I think this could have been closed earlier as this issue seems to be about figuring out whether and how to correct for gravity. We know this now.

But the code in ESSreflectometry is still separate from ESSsans and ScippNeutron. We can, and I think should, merge the implementations. I opened a new issue in ESSreflectometry to track this: https://github.com/scipp/essreflectometry/issues/54

closed 2024-06-24 by jl-wynen