scipp/scippneutron #99: scippneutron install issue.

fott, 2021-06-11:
Dear developers,

I have just discovered the existence of "scippneutron".

scipp 0.7.0 is installed (via conda) and works nicely.

When I follow the installation procedure for "scippneutron" (with the environment file)
conda env create -f scippneutron.yml

I am getting the message:
Collecting package metadata (repodata.json): done
Solving environment: failed
ResolvePackageNotFound:
  - scipp/label/dev::mantid-framework
  - python-confluent-kafka
 
(the packages are indeed not available on conda. 
conda search finds no result, even on the scipp/win-64 channel)

As an "end-user", I cannot go further in the investigation.

Best Regards

nvaytet, 2021-06-14:
Hi there,

Thanks for your interest in `scippneutron`.

Unfortunately, due to technical difficulties, we do not currently have a `mantid-framework` conda package for Windows.
Only linux and osx are supported for now.
We use the `mantid-framework` to load neutron data files from facilities outside of ESS.

You can however install and use `scippneutron` without having the `mantid-framework`, by using this `.yml` file instead: https://github.com/scipp/scippneutron/blob/main/scippneutron-developer-no-mantid.yml (or the raw file https://raw.githubusercontent.com/scipp/scippneutron/main/scippneutron-developer-no-mantid.yml).

The functionality will be however relatively limited, if you are looking to work with neutron data files from existing facilities.
If you are tied to Windows, another option would be to use the Windows Subsystem for Linux (https://docs.microsoft.com/en-us/windows/wsl/about) to use `scippneutron`. You would set up a linux environment that way on your machine, and then install `scippneutron` via conda there.
Let us know if you need further help in getting this set up.

fott, 2021-06-15:
Thank you, working OK.
It was just unexpected that <scippneutron> and <jupyterlab> are not explicitely in the YAML file (could be installed "manually" though)

For your information,
We are currently setting up a new TOF instrument on a new source and we want to "by-pass" the Mantid step. Hence we are not interested in any legacy file format.

Thank you for the fast development pace of scipp

_______________________________
Frédéric OTT
Lab. Léon Brillouin CEA/CNRS, bat 563
Centre d'Etudes de Saclay
91191 Gif sur Yvette Cedex FRANCE
Tel : 33 1 69 08 61 21 / 65 16
Fax : 33 1 69 08 82 61
E-mail : ***@***.******@***.***>
Projets IPHI-Neutrons et SONATE<http://iramis.cea.fr/llb/Phocea/Vie_des_labos/Ast/ast_sstechnique.php?id_ast=2755&voir=2954>

OwenArnold, 2021-06-15:
@fott  Interesting. What is your instrument exactly?

Yes you can add what you need to the yaml files, or `conda install` your other dependencies. With the latter just be careful about downgraded runtime dependencies, I'm not aware this would be a problem for `jupyter` but it might be for other things you need.

fott, 2021-06-15:
Dear Owen,

We have built a small accelerator based neutron source at Saclay (IPHI - Neutrons).
We have moved parts of a powder diffractrometer from the Orphée reactor to this source to build a new "Generic Scattering Spectrometers" (DIoGENE) to evaluate the performances of the small source.

We are considering different options for the TOF data acquisition and processing.
scipp is currenty our preferred choice for its simplicity and easy of integration in python programs

Best Regards

_______________________________
Frédéric OTT
Lab. Léon Brillouin CEA/CNRS, bat 563
Centre d'Etudes de Saclay
91191 Gif sur Yvette Cedex FRANCE
Tel : 33 1 69 08 61 21 / 65 16
Fax : 33 1 69 08 82 61
E-mail : ***@***.******@***.***>
Projets IPHI-Neutrons et SONATE<http://iramis.cea.fr/llb/Phocea/Vie_des_labos/Ast/ast_sstechnique.php?id_ast=2755&voir=2954>

OwenArnold, 2021-06-15:
> We are considering different options for the TOF data acquisition and processing. scipp is currenty our preferred choice for its simplicity and easy of integration in python programs

Great. We look forward to hearing more from you in that case!

closed 2021-07-21 by SimonHeybrock