scipp/scippneutron #132: Each filter needs a test - create a list of filters that need testing and bench-marking

Pasarus, 2021-08-16:


Pasarus, 2021-08-16:
Mantid event filters:
- CalculateCountRate
- CreateLogTimeCorrection
- ExportTimeSeriesLog
- FilterBadPulses
- FilterByLogValue
- FilterByTime
- FilterByXValue
- FilterEvents
- FilterLogByTime
- GenerateEventsFilter
- GetTimeSeriesLogInformation
- RebinByTimeAtSample

Possibles:
- SortEvents
- CorelliCrossCorrelate

SimonHeybrock, 2021-08-18:
> SortEvents
>   Sort the events in an EventWorkspace

Note that none if the scipp algorithms rely on sorted events. This needs to be taken into account since Mantid keeps the sort state internally, so algorithm executions that require sorting show vastly different performance depending on what was called before, or whether the algorithm is called twice on the same workspace. In scipp there is no such inconsistency.

Pasarus, 2021-08-19:
Thank you for that @SimonHeybrock I will bare that in mind when attempting to benchmark, it is frustrating isolating features sometimes.

closed 2021-11-15 by jl-wynen