scipp/scippneutron #141: Testing and Design for low-level Nexus utilities

OwenArnold, 2021-08-20:
To avoid the mistakes that accompanied growth of the mantid loaders, we should consider restructuring some parts of the nexus loading utilities. To support the expanded information obtainable from Nexus the loaders will need to do more and we should pause to consider this.

For example,  I think it would be important to introduce testing for the low-level concepts and building blocks such as the methods of `LoadFromHdf5` which are used as utilities onto which the nexus building is structure but could easily disintegrate into quite a mess.

SimonHeybrock, 2021-08-20:
> For example, I think it would be important to introduce testing for the low-level concepts and building blocks such as the methods of LoadFromHdf5 which are used as utilities

Not sure exactly what you mean, but I feel that implementation details should *not* be tested, since it often makes refactoring harder. Tests should mostly focus on the contract of the user-facing functionality.

I agree though that more components that are usable (and test-worthy) could be useful. For example, we may want to support stand-alone loading of individual Nexus classes such as `NXevent_data` or `NXsample`. That would be both useful and provide a smaller feature that can be tested more easily.

OwenArnold, 2021-08-20:
> > For example, we may want to support stand-alone loading of individual Nexus classes such as NXevent_data or NXsample. That would be both useful and provide a smaller feature (and building blocks) that can be tested more easily.

This is essentially what I was trying to convey above.

PR #260 (2022-01-27): Extend nexus loading utilities, add NXdata and NXdetector
There are a number of question marks attached to some of the functionality, pending answers on how to interpret the Nexus standard.

- Closes #257.
- Closes #226, at least for the basics. No support in `load_nexus` added, but this might not be what we want anyway.
- Closes #141, even though we may want to wrap things up further in the future. For now this should be a start and enables users to load individual components without forcing them into specific transformations and assembly of pieces from a file.
closed 2022-01-27 by SimonHeybrock