scipp/scippneutron #219: Cover all Mantid event filters

jl-wynen, 2021-11-15:
Mantid event filters:
- FilterBadPulses
    - Filters out events associated with pulses that happen when proton charge is lower than a given percentage of the average
- FilterByLogValue
    - Filter out events from an EventWorkspace based on a sample log value satisfying filter criteria
- FilterByTime
    - This algorithm filters out events from an EventWorkspace that are not between given start and stop times
- FilterByXValue
    - Filters events according to a min and/or max value of X
- FilterEvents
    - Filter events from an EventWorkspace to one or multiple EventWorkspaces according to a series of splitters.
- FilterLogByTime
    - Filters a log between time intervals and applies a user defined operation to the result
- SortEvents
    - Sort the events in an EventWorkspace
- GenerateEventsFilter
    - Generate one or a set of event filters according to time or specified log’s value.

Possibles:
- RebinByTimeAtSample **maybe**
    - Rebins with an x-axis of relative time at sample for comparing event arrival time at the sample environment
- FilterEventsByLogValuePreNexus
    - Load and split SNS raw neutron event data format and stores it in a workspace

_Originally posted by @Pasarus in https://github.com/scipp/scippneutron/issues/132#issuecomment-899402496_

SimonHeybrock, 2022-10-28:
I suggest closing this as I see little utility. We can address missing cases if/when they arise over time based on user requests.

jl-wynen, 2022-10-28:
Should we add a how to in the scippneutron docs? Or do you think that the scipp docs page is sufficient?

SimonHeybrock, 2022-10-28:
The scipp docs page is actually using neutron data, so for now I would think it is sufficient.

In the past we had considered having detailed pages showing how Mantid features can be achieved with Scipp, but by now I think this would be a waste of time and take away the focus from proper documentation.

closed 2022-10-28 by jl-wynen