scipp/scippneutron #221: Names of file loaders

jl-wynen, 2021-11-15:
- Should `scn.load` be renamed to be explicit about using mantid?
- Inconsistent keyword args

SimonHeybrock, 2022-10-28:
> Should scn.load be renamed to be explicit about using mantid?

We should do that soon, and add a deprecation warning to the old one.

jl-wynen, 2023-03-20:
Let's finally do this and rename `load` to `load_mantid` and deprecate `load` for 6 months!

Unfortunately, I don't remember what I mean by 
> Inconsistent keyword args

SimonHeybrock, 2023-03-20:
`load_mantid` may be slightly misleading, since it is not (just) for loading Mantid files?

jl-wynen, 2023-03-20:
`load_with_mantid`, `load_via_mantid`, `mantid_load`?

jl-wynen, 2023-03-20:
Or move it into the mantid module and don't import at top level. So users can use `scn.mantid.load`.

SimonHeybrock, 2023-03-20:
Submodule sounds good 👍

SimonHeybrock, 2023-03-20:
On the other hand... we may want to add scippnexus-based loaders, and we will not name them `scn.scippnexus.load_abc`? So maybe `load_with_mantid` is best?

PR #436 (2023-04-19): Rename load to load_with_mantid
Fixes  #221
closed 2023-04-19 by jl-wynen