scipp/scippneutron #237: Event filtering toolkit

SimonHeybrock, 2021-12-16:
https://github.com/scipp/ess/pull/57 tried to implement Mantid event-filtering functionality based on scipp. After several review rounds and small improvements in scipp itself, little code is left (apart from unit tests). This thread is meant to establish how we can move forward. My comments refer to [filter.py](https://github.com/scipp/ess/pull/57/files#diff-460ee6d77bed9b6d716f5c5d70c48d9c6b99de4158d4e90e40b0ae035f92bd2b). The PR adds the following (ignoring parts that are now redundant):

1. `filter_by_proton_charge`: There are two things done here:
   - Renaming of the time-series log dimension. I believe this is an artifact of how we load data, and should *not* be part of a filtering function. Can we fix our loaders to use the same dimension name? See also #236.
   - Defining a simple condition. If we ignore the upper limit this simply does `filter(da, proton_charge > 0.9 * proton_charge.mean()`, and a dedicated function does not appear useful.
   
   Aside from that, nothing is special about "proton charge", and *if* a interval-based filter *was* added, it could be generic, which would be more inline with scipp's mentality.
2. `filter`: With recent improvements to `sc.lookup`, finding edges is not necessary any more. What remains is:
   - Add new coord (to shallow copy) of input data array, defined using `sc.lookup()` using the input condition
   - Define groups to request. This could be unique values of input condition.
   - Call `sc.bin` with desired groups
   - Return result, or slice of result (depending on whether all or just a single interval is needed)
3. `filter_by_value`: Extract points that match a condition, e.g., range of values. This just uses `groupby`, but wraps code needed to temporarily add condition as coord, extracting the desired group, and removing the temporary coord. While not clear exactly how, it looks like this could be supported by slight improvements to the upstream scipp API?

Item 2.) is likely the most relevant contribution. At the same time the interface as it is, is too limiting. We have a combination of cases:

- Extract single condition value or multiple values (as extra dim)?
- Extract based on exact values (groups) or intervals (edges)?

This leads us to 4 cases in total. Is there a single function that can support this with a sensible API, or do we need multiple? Do we need one at all, or would making the individual steps (lookup to define new coord, defining groups) more simple be sufficient, such that users can call `sc.bin` directly?

SimonHeybrock, 2021-12-16:
Regarding item 2.), let me give a couple of examples of the code this `filter` function would replace:

### Examples

#### Case a): Extract single value

```python
tmp = da.copy(deep=False)  # optional, may if we want to keep input clean
tmp.coords['good_pulse'] = sc.lookup(good_pulse)[da.bins.coords[good_pulse.dim]]
filtered = sc.bin(tmp, groups=[sc.array(dims=['good_pulse'], values=[True])])
# optional: get rid of extra dim
filtered = filtered['good_pulse', 0]
```

#### Case b): Group by values

```python
tmp = da.copy(deep=False)  # optional, may if we want to keep input clean
tmp.coords['good_pulse'] = sc.lookup(good_pulse)[da.bins.coords[good_pulse.dim]]
filtered = sc.bin(tmp, groups=[sc.array(dims=['good_pulse'], values=[False, True])])
```

#### Case c): Extract interval

```python
tmp = da.copy(deep=False)  # optional, may if we want to keep input clean
tmp.coords['temperature'] = sc.lookup(temperature)[da.bins.coords[temperature.dim]]
filtered = sc.bin(tmp, edges=[sc.array(dims=['temperature'], unit='K', values=[100.0, 120.0])])
# optional: get rid of extra dim
filtered = filtered['temperature', 0]
```

#### Case d): Group into intervals

```python
tmp = da.copy(deep=False)  # optional, may if we want to keep input clean
tmp.coords['temperature'] = sc.lookup(temperature)[da.bins.coords[temperature.dim]]
filtered = sc.bin(tmp, edges=[sc.linspace(dim='temperature', unit='K', start=100.0, stop=120.0, num=11)])
```

### Discussion

- It is hard to imagine a function signature that preserves flexibility but does not become too complicated.
- In practice this is little more than two lines. In many cases the user will need to define desired groups/edges anyway.

SimonHeybrock, 2021-12-17:
With the suggestion of `generate_coord`, the full example for filtering bad pulses would turn into:

```python
proton_charge = da.attrs['proton_charge'].value
# assuming `proton_charge` is a histogram, may need extra step here
da = generate_coord(da, good_pulse=proton_charge > 0.9*proton_charge.mean())
good_pulse = sc.array(dims=['good_pulse'], values=[True])
da = sc.bin(da, group=[good_pulse])['good_pulse', 0]
```

SimonHeybrock, 2022-01-03:
Assuming some upstream convenience is added:
```python
proton_charge = da.attrs['proton_charge'].value
da = generate_coord(da, good_pulse=proton_charge > 0.9*proton_charge.mean())
da = sc.bin(da, group=[sc.array(good_pulse=[True])]).squeeze('good_pulse')
```

SimonHeybrock, 2022-01-03:
Another example:

```python
temp = da.attrs['sample_temperature'].value
da = generate_coord(da, temperature=temp)
da = sc.bin(da, edges=[sc.array(temperature=[4.2, 4.3], unit='K')]).squeeze('temperature')
```

jl-wynen, 2022-03-15:
Additional complications: (discussion with Simon and Neil)

The coord of the log to filter by needs to be converted to bin-edges in order to use `lookup`. This needs knowledge about how the data was recorded:
- does the time in the log encode start times for bins or bin centres, or something else?
- how does the pulse time of the log (e.g. proton charge) relate to the time of the events in the detector?
- the bins of the log might have to be grown to be sure to filter out all bad data. (Difficult for strongly oscillating log)

These questions cannot be answered in general; the user needs to make decisions. So we should provide an API that helps specify these things.

We should have a generic filtering function and potentially wrappers for specific filters.
We could also try a two step approach like in groupby where the user first designs a filter using the above parameters and then applies it. This makes the two components more testable.

SimonHeybrock, 2022-06-22:
We should probably aim for something that can be used with `transform_coords`? Then, using the upcoming API in scipp, we can:

```python
da.transform_coords('temperature', graph={'temperature':sc.lookup(temp)})
  .bin(temperature=10)
  .hist()
  .plot()
```

Aside: I think if we want to truly make `transform_coords` useful for every day use, we need to provide a simpler syntax (avoiding repetition) for the common case of making, e.g., a single new coord. Could we support supplying a positional dict, in which case all dict keys are computed as outputs?

```python
da.transform_coords({'temperature':sc.lookup(temp)})
```
Or even support keywords? I know this may clash with our "reserved" named arguments, but what are the odds in practice?

```python
da.transform_coords(temperature=sc.lookup(temp))
```

SimonHeybrock, 2022-06-22:
Regarding `sc.lookup`: This works fine of the time-series-log used for lookup has discrete values. If it is some noisy sensor such as temperature we need to "discretize" this first. This could be done by hand, but we may also consider providing an option in `sc.lookup` (which may actually be a class of similar functions?):

```python
sc.lookup(temperature, tolerance=0.1 * sc.Unit('K'))
```

SimonHeybrock, 2022-07-05:
Note that the simpler `transform_coords` API essentially turns this into the `generate_coord` function proposed earlier. What remains to be done is to come up with a slightly more flexible solution along the lines of `sc.lookup` that is compatible with `transform_coords`.

SimonHeybrock, 2022-07-05:
Status:

- [ ] scipp/scipp#2670 should simplify parts of this
- [ ] scipp/scipp#2674 aims to make `sc.lookup` into something slightly more generic that can be used with `transform_coords`
- [ ] Once done, check status of cases. Is everything simple enough?
- [ ] Add documentation with usage examples.

SimonHeybrock, 2022-07-05:
Revisiting the examples provided earlier, based on recent/upcoming improvements in scipp:

#### Case a): Extract single value

```python
filtered = da.transform_coords(good_pulse=sc.lookup(good_pulse)) \
             .group('good_pulse')['good_pulse', sc.index(True)]
```

#### Case b): Group by values

```python
tmp = da.copy(deep=False)  # optional, if we want to keep input clean
tmp.bins.coords['good_pulse'] = sc.lookup(good_pulse)[da.bins.coords[good_pulse.dim]]
filtered = sc.bin(tmp, groups=[sc.array(dims=['good_pulse'], values=[False, True])])
```

#### Case c): Extract interval

```python
filtered = da.transform_coords(temperature=sc.lookup(temperature)) \
             .bin(temperature=sc.array(dims=['temperature'], unit='K', values=[100.0, 120.0])) \
             .squeeze('temperature')  # optional: get rid of extra dim
```

#### Case d): Group into intervals

```python
filtered = da.transform_coords(temperature=sc.lookup(temperature)) \
             .bin(temperature=10)  # alternative: provide explicit interval

jl-wynen, 2022-08-26:
Is everything we need supported by scipp now? If so, we should provide cookbook-style docs for filtering and potentially wrap some of them in dedicated functions to avoid code duplication across workflows.

See also a more complete list: #219 Are we missing anything from that?

SimonHeybrock, 2022-08-26:
See checklist above in https://github.com/scipp/scippneutron/issues/237#issuecomment-1174864904

SimonHeybrock, 2022-08-26:
> See also a more complete list: https://github.com/scipp/scippneutron/issues/219 Are we missing anything from that?

I don't think we miss anything important.

SimonHeybrock, 2022-10-28:
Scipp 0.17 added the "final" features for this as well as documentation (https://scipp.github.io/user-guide/binned-data/filtering.html).

closed 2022-10-28 by SimonHeybrock