scipp/scippneutron #238: Histogram and bin-edge toolkit

SimonHeybrock, 2021-12-17:
There are a number of small operations that are semi-frequently needed when operating on data or meta data:

1. bin edges <-> bin centers
2. histogram <-> "point data" (mostly same as 1.)
3. add trailing edge to time series, if only interval starts and values are stored (alternatively: trim last data value since end time not known).

A problem with most of the above is that they are not clearly defined and implementation (such as in Mantid) "guess" a reasonable default assumption. For example, conversion of bin centers to edges uses a symmetric definition, which is correct only for constant bin widths.

nvaytet, 2022-01-19:
Maybe we could just make it throw if bin widths are not constant? It seems constant bin widths will be the most common case anyway?

SimonHeybrock, 2022-01-19:
Well, there is constant and *constant*. If we do that we will get issues with threshold of what should be accepted, etc.

Is there big harm in just supporting everything when computing midpoints?

nvaytet, 2022-01-19:
I think there is no harm in computing mid-points.
For converting from midpoints to edges, i would say it's only ill-defined at the boundaries, since all the other edges are computed using midpoints offset by 1.
As long as we document the behaviour at the boundaries properly, I think it's ok?
Better than not providing the functionality to the users at all, in my opinion.

SimonHeybrock, 2022-01-19:
Not sure how much demand there actually is for converting to "bin edges". I'd suggest we only add `midpoints` for now?

SimonHeybrock, 2022-01-20:
Decision:
- Add `midpoints` now.
- Postpone decision on others.

nvaytet, 2022-02-01:
Can we close this?

SimonHeybrock, 2022-02-02:
👍 (and open a new one once we decide we need more)

closed 2022-02-02 by SimonHeybrock