scipp/scippneutron #263: Builder for Nexus loaders

SimonHeybrock, 2022-01-31:
`load_nexus` and other loading facilities apply a number of transformations when loading data. Examples include:

- #236
- #34

It is often instrument-dependent which transformations can and should be applied. Furthermore, it may be required to *configure* the transformation, e.g., whether to keep input coords or not. A builder class could be a good and flexibly solution:

```python
class NexusLoaderBuilder:
    def add_coord_transform(self, ...)  # same interface as `transform_coords`
    
builder = NexusLoaderBuilder()
builder.add_coord_transform('position', graph=scn.nexus.compute_positions())
builder.add_coord_transform('time_of_flight', graph=scn.nexus.make_tof_frames())
loader = builder.build()

da = loader(filename)
```

We could, e.g., define pre-configured loader for a specific instrument.

jl-wynen, 2022-01-31:
> The third item (binning into pixels) does not fall into the pattern, since it is not a coord transform. Not clear what the best solution for that is.

The builder can also have different methods, not everything needs to be a coord transform.

SimonHeybrock, 2022-02-02:
One thing I forgot to discuss above is that certain transformations must be applied on a per-detector (NXdetector or NXevent_data) basis. In particular the spatial transforms that @Tom-Willemsen is working on cannot be done after the current concatenation of banks that `load_nexus` does automatically (unless we want to broadcast the transforms, which is probably not desirable).

SimonHeybrock, 2022-03-24:
It is becoming clearer that an approach like this has to have a wider scope than described in the original comment. We need to be able to do the following:

- Configure groups to be included (by name, or possibly by NX_class).
- Configure groups to be excluded (by name, or possibly by NX_class).
- Configure processing/transformation steps for groups by NX_class, e.g., processing of an NXdetector.
- Configure structure of output, including:
  - Placement of specific NX_classes, e.g., to store all NXmonitor as attributes.
  - How to represent data. We may, e.g., need to merge multiple NXdetector groups such that they can be stored in a single DataArray (or not, provide different options).
  - Placement of specific meta data by name. For example, auxiliary optical spectrometer data may be stored as an attribute. Sample and source positions need to be stored as coordinates.
- Final transformations on the entire output.

jl-wynen, 2022-03-24:
Would a configurable collection of classes be a better fit for this task?
e.g. an association via the class
```.py
class NXDataDREAM(NXData):
    def __getitem__(self, name):
        x = super()[name]
        # process x

loader = nexus.Loader()
loader[NXclass.NX_data] = NXDataDREAM
```

SimonHeybrock, 2022-03-24:
It could be a solution for one of the bullets above, but cannot handle the others, I think?

SimonHeybrock, 2022-03-24:
Note that I am not suggesting restricting the architecture to a builder pattern, that was just the initial idea.

jl-wynen, 2022-03-24:
I think it can handle all bullets except the varying file structure which might be challenging. The strength is that you don't have to call `super().__getitem__`

SimonHeybrock, 2022-03-24:
Not sure I understand, we best discuss this on a whiteboard.

SimonHeybrock, 2022-04-12:
Notes from discussion with @jl-wynen:

```python
import scippnexus as snx

loader = DataArrayLoader()  # class for configuring a load function the produces DataArray

# add require monitors with known names
loader.attrs['bank1'] = delayed(sc.scalar)(loader.get(NX_class.NXdetector, 'bank1'))
loader.attrs['monitor1'] = delayed(sc.scalar)(loader.get(NX_class.NXmonitor, 'monitor1'))
loader.attrs['monitor2'] = delayed(sc.scalar)(loader.get(NX_class.NXmonitor, 'monitor2'))
loader.attrs['instrument_name'] = delayed(identity)(f['instrument']['instrument_name'][...])

# how to add "all logs", we don' know until we open a concrete file!
def pick(group: snx.NXobject):
    if group.nx_class != NX_class.NXlog:
        return False
    if group.name.contains(pattern):
        return False
    return True
    
# need to be able to specifiy arbitrary pre-processing steps (before adding as attr)
loader.set_dynamic_attrs(identity,
                         parent=snx.NX_class.NXinstrument,
                         names=['instrument_name', 'facility', 'run_start'])
loader.set_dynamic_attrs(sc.scalar, nxclass=snx.NX_class.NXlog)  # all
loader.set_dynamic_attrs(sc.scalar, nxclass=snx.NX_class.NXlog, exclude=['large-log'])
loader.set_dynamic_attrs(sc.scalar, nxclass=snx.NX_class.NXlog, parent=['instrument'])
loader.set_dynamic_attrs(sc.scalar, by_class='NXentry.NXinstrument.NXlog')  # all logs in these classes, name ignored

loader.set_dynamic_attrs(sc.scalar, snx.NX_class.NXdisk_chopper, pick_chopper)
loader.attrs[name] = delayed(sc.scalar)(loader.get(NX_class.NXlog, name))

loader.coords['sample_position'] = (get_position, loader.get(NX_class.NXsample))
loader.coords['source_position'] = (get_position, loader.get(NX_class.NXsource))

loader.attrs['bank1'] = delayed(sc.scalar)(loader.get(NX_class.NXdetector, 'bank1'), filter=select_events)
#load = loader.make_load_function()

da1 = loader(filename1)  # use for multiple files
da2 = loader(filename2)
da3 = loader(filename2, select_events=('pulse', slice(0,0)))
```
```python
# want to be able to load only meta data, or data array with equivalent structure?
# - easy for events, just set bin sizes to 0
# - what about, e.g., large monitors? would also be events

# once we loaded meta data, how can we avoid reloading it once we load full file?
da = load(filename1, pulse_max=1000)
da.data = loader.data  # loads actual data
da.attrs['monitor1'] = loader.attrs['monitor1']
da.attrs['large-log'] = loader.attrs['large-log']  # previously excluded

da.data = load_events(filename1)

myloader.attrs['monitor1'] = (sc.scalar, ...)
myloader.attrs['temperature'] = sc.scalar(1.2, unit='K')
myloader.append_filter(transform_coords)
myloader.append_filter(rename_dims)
loader = myloader(filename)
loader['pixel', :1000]  # returns DataArray, binned by pixel
loader.attrs['monitor2'] = (sc.scalar, ...)  # is it feasible to add new attrs later, or does it make caching mechanism too complicated?
loader['pixel', :2000]  # can return with already loaded meta data (and pixel chunks)
loader.select_events['pulse', :1000]['pixel', :]  # load no events
loader.select_events['pulse', :]['pixel', :]  # implementation detail: may use cached events
loader['pixel', :]  # load everything
# can we modify existing loader instance if we want to, e.g., add loading of more meta data?

# merge information from multiple files?
```

Will make a prototype based on this.

SimonHeybrock, 2022-04-20:
Some experiments on how to setup loading before opening a file. This examples defines steps to take for loading logs into a dict:

```python
import scipp as sc
import scippnexus as snx
import scippnexus.data
filename = snx.data.get_path('PG3_4844_event.nxs')
f = snx.File(filename)
import dask

@dask.delayed
def load_or_none(nxobject):
    try:
        if 'Speed' in nxobject.name:
            raise RuntimeError('xxx')
        print(f'Loading {nxobject.name}')
        # May want to use from_nxobject instead of this
        return nxobject[...]
    except RuntimeError:
        return None
    
def load_all(objects):
    dsk = {obj.name:load_or_none(obj) for obj in objects}
    dsk = exclude(dsk, 'Chopper')
    dsk.pop('/entry/DASlogs/PhaseRequest3', None)
    dsk.pop('/entry/DASlogs/PhaseRequest4', None)  # does not exist
    dsk['xxx'] = dask.delayed(sc.sum)(dsk['/entry/DASlogs/proton_charge'])
    return dsk

def exclude(dsk, pattern):
    return {k:v for k,v in dsk.items() if pattern not in k}
    
dsk = load_all(f.by_nx_class()[snx.NX_class.NXlog].values())

dask.visualize(dsk, optimize_graph=True)
dask.compute(dsk)
```

This demonstrates:
- Setting up filters
- Setting up processing steps
- Dealing with things that fail to load/process (return None)

The last item is the only not-so-nice one: Downstream code will have to deal with dicts that contain `None` values... but maybe this is good, it allows for distinction of things that are there/expected from things not are missing.

TODO:
- Can the plain `nxobject[...]` be replaced by a collection, i.e., we operate on a dict of collections?

SimonHeybrock, 2022-09-08:
Closing, as scipp/ess#139 provides a different approach for the same problem.

closed 2022-09-08 by SimonHeybrock