scipp/scippneutron #341: Dashboard / live data considerations

SimonHeybrock, 2022-06-21:
Related: #291, in particular:

[...] the consensus is that the current requirements for data streaming are too fuzzy and maybe too ambitious. It appears to stop us from make actual progress. Therefore:

- Aim for a minimal working but useful solution.
  - Must get away from the "we have a working prototype" situation as soon as possible. 
- Should be trivial to launch, e.g., based on config file, without complicated setup.
  - Integration tests to ensure this keeps working over the coming years.
- Minimal features:
  - Small number of pre-configured live-updating plots (instrument view, normal plots).

_Originally posted by @SimonHeybrock in https://github.com/scipp/scippneutron/issues/291#issuecomment-1119441599_

Further considerations:

- I think we should consider avoiding coupling to the `scippneutron` and/or `ess` packages (and Mantid). `scippneutron` is compiled and appears to be trickier to keep compatible / release than the pure Python `scippnexus`. For dashboards I imagine that there will be a lot of iteration so the ability to release quickly would be an advantage.
  - Note that this does *not* imply that we cannot *depend* on, e.g., `scippneutron` (but without a `pip` package that would be a bit annoying).
- The dashboard could be its own library (depending possibly only on `scipp`). This could then be used, e.g., by `ess` to compose it with a Kafka listener (currently in `scippneutron`, but I suppose this could be moved to a stand-alone package if advantageous).
  - Does this make any sense, or are is the dashboard code so lightweight and instrument-dependent that it should just live in `ess`?

jl-wynen, 2022-06-21:
How would we use the instrument view if dashboards don't depend on scippneutron?

We will have to iterate quickly on ess, too. So we could put the dashboards there. I doubt there will be a lot of code that other facilities could / want to use.

SimonHeybrock, 2022-07-18:
> How would we use the instrument view if dashboards don't depend on scippneutron?

The underlying plot is actually the same. `scippneutron` "only" provides a wrapper for adding some extra elements, and more convenient setup. It may thus be possible to avoid the dependency, but later still use the dashboard with the instrument view (instead of the underlying 3d scatter plot).

SimonHeybrock, 2023-04-18:
See `beamlime`

closed 2023-04-18 by SimonHeybrock