scipp/scippneutron #359: Future of `load_nexus`

SimonHeybrock, 2022-08-01:
#263 is currently stuck, and the considered approaches may be too complex to be worthwhile. There are a couple of things that should be fixed/changed independently of that. This could be a behavior change of `load_nexus`, or an additional function (and deprecate `load_nexus`?).

Items:

- Do not enforce "canonical" (and often wrong) coord/dim names. #225
- Do not add a TOF dimension of length one. This will not be useful any more with the next plotting implementation (on the contrary, it will get in the way).
- #354 could simplify the code.
- Do not `flatten` banks, if possible. Not clear what to do instead. If shapes are compatible we can concat in extra dim, but in general this is not possible. This also links to handling of meta data (see next item).
- "Flattening" meta data into the `attrs` dict (wrapped in scalar variable) is cumbersome. Consider returning a more structured object that preserved some of the Nexus structure. For example, we might have `obj.detector['bank11']`,  `obj.monitor['transmission']`, and `obj.log['sample_temperature']`.

The final two items show that Scipp's `DataArray` and `Dataset` or too simple/limited to fully support the real structure of the data. Dim/coord alignment in `Dataset` may be circumvented by using distinct dim names (`bank1_pixel`, `bank2_pixel`) if they are not aligned. This does however not solve the problem entirely. I do not necessarily see this as a sign that we need to support more complex data structures in Scipp. Instead, it may simply lead to the realization that downstream packages (such as `scippneutron` and `ess`) may want to work with more neutron-specific structures, built around `scipp.DataArray`. A plain dict or a data class may be a starting point. Can we be better at supporting mapping scipp operations over dicts or lists? We may, e.g., want to apply the same processing to all detector banks.

SimonHeybrock, 2022-08-01:
Started some experiments in https://github.com/scipp/scippneutron/compare/nexus-load-as-is?expand=1. As you can see for now this is just 20 lines. I think we will need to provide some of the transforms that `load_nexus` does as helper functions, so they can be applied/configured individually as required. Probably this should be in a submodule? We should also consider splitting the structure setup and the loading. This would let the user apply things such as pulse-time selection to load only parts, or to delete entries.

jl-wynen, 2022-08-02:
Is your proposed data structure fundamentally different from attrs in a data array? The various dicts could be flattened into a single one by using keys like `monitor:transmission` and `log:sample_temperature`. And the different detectors could be returned as a data array each.
I'm not suggesting to do this. But what do we gain by having only a single dict for e.g. all logs? It doesn't preserve the structure in the file.

SimonHeybrock, 2022-08-02:
> Is your proposed data structure fundamentally different from attrs in a data array?

It also avoids wrapping in a scalar.

> But what do we gain by having only a single dict for e.g. all logs? It doesn't preserve the structure in the file.

Good point. *Usually* it is only a single one, but this aspect may indeed require a little more thought. My main problem is that the Nexus structure is not very "sensible" to a user, so I wanted to make it somewhat simpler.

SimonHeybrock, 2022-08-02:
Maybe keeping the full structure would be simplest? Then we have to worry about the least number of things here. If we do that then this should probably move into ScippNexus.

SimonHeybrock, 2022-08-02:
Basically, this would translate a read-only Nexus structure (provided by scippnexus), into something editable that terminates at the level where scippnexus can load an object (group or dataset):

Nexus file:
```
NXentry
  title
  NXsample
    ...
  NXmonitor
    ...
  NXinstrument
    NXdetector
      NXevent_data
        ...
    NXdetector
      ...
    NXlog
      ...
    NXlog
      ...
```

Loaded (and intermediate) structure:
```
entry: dict
  title: sc.Variable
  sample: sc.Dataset
  monitor: sc.DataArray
  instrument: dict
    detector1: sc.DataArray
    detector2: sc.DataArray
    log1: sc.DataArray
    log2: sc.DataArray
```
Basically, we replace tree branches by the corresponding `scippnexus.NXobject` as soon as it is "loadable" (scippnexus cannot load `NXinstrument`, but it can load `NXdetector`, including nested `NXevent_data`). We first have an intermediate structure (nested dict of `scippnexus.NXobjects`) which can be "edited" (remove entries, etc.). This can then be loaded, recursively.

SimonHeybrock, 2022-08-02:
Immediate problem: The above works if we just use string keys, but we "lose" the NX_class. Should the NX_class be used to provide special properties for accessing this? That is, should we support access without name?

```python
data.entry.instrument.detector['detector1']  # this??
```

How do deal with groups of which there could be more than one? Is there always just one instrument? One sample?

SimonHeybrock, 2022-08-02:
We could to the "usual" plural/singular thing:

```python
data.entry.instrument.detector  # single entry, single detector
data.entry  # raise if multiple
data.entries['entry1'].instrument.detectors['bank11']
data.entry.sample  # normal, single sample
data.entry.samples['sample1']  # more special case, multiple samples
```

nvaytet, 2022-08-02:
So
```
data.entry.instrument.detector['detector1']
```
would just be a shorthand for
```
data['entry']['instrument']['detector']['detector1']
```
?
Not sure it's so much shorter? But maybe nice for auto-completion?

SimonHeybrock, 2022-08-02:
> Or are entry and instrument special names you wanted to have to mirror NXentry and NXinstrument?

Yes, exactly.

It is much shorter if you consider that the actual keys could be anything, and are often meaningless or unknown to the user. For example, ISIS uses `raw_data1` for their "entry", i.e., it would be:

```python
data['raw_data1']
```

Furthermore, by using only the string key we lose the validation of "getting what you ask for". With a dedicated property you ensure that you actually *do* get an entry/instrument/sample, not something that happens to match the name. It would also make the code more self-documenting. Instead of properties, another option would be to add the NX_class as part of the dict key:

```python
data[NX_class.NXentry, 'raw_data1']
data[NX_class.NXmonitor, 'monitor1']
```

This makes this code safe and self-documenting, but still requires knowing the string key, which would be nice to avoid.

nvaytet, 2022-08-02:
Ok, so `data['raw_data1']` would just be `data.entry`. I get it now.

For the singular/plural issue, would it be possible to have something where `data.entry.instrument` is possible if only one `entry` exists, but you would have to do `data.entry['entry1'].instrument` if there are multiple? So `data.entry` would be some sort of dict object that has direct accessors if there is only a single key?

Maybe it's a little weird, but I feel the singular/plural tends to confuse users (a bit, though not directly related, like the `.value/.values`)

SimonHeybrock, 2022-08-02:
> For the singular/plural issue, would it be possible to have something where `data.entry.instrument` is possible if only one `entry` exists, but you would have to do `data.entry['entry1'].instrument` [...]

I had considered that, but wouldn't it lead to very confusing errors?

Maybe we can combine the tuple-keys with the singular accessors?

```python
data.entry  # gives data[NX_class.NXentry, xxx] if there is only a single key with NXentry
data[NX_class.NXentry, 'entry1']  # must use this if multiple entries
```

SimonHeybrock, 2022-08-02:
I think ultimately we want something a bit like `collections.namedtuple`, with two orthogonal naming systems. Not sure that helps.

nvaytet, 2022-08-02:
> I had considered that, but wouldn't it lead to very confusing errors?

What kind of errors are you thinking about?

```Py
data.entry.instrument
> data.entry has no attribute "instrument"
```
?

SimonHeybrock, 2022-08-02:
> > I had considered that, but wouldn't it lead to very confusing errors?
> 
> What kind of errors are you thinking about?
> 
> ```python
> data.entry.instrument
> > data.entry has no attribute "instrument"
> ```
> 
> ?

```python
data.entry['sample1']
```
User wants to access `sample1` within the entry, but `entry` gave a list of entries.

SimonHeybrock, 2022-08-23:
The plural/singular approach has a fundamental issue: Some of the NX_class names are plural by default, e.g., `NXreflections` or `NXtransformations`. We need to find a different approach.

SimonHeybrock, 2022-08-23:
On possible option would be to enhance `__getitem__` to accept an NX_class, to return a list of all matches. The properties would only work if there is a unique match:

```
f.entry
f.entry.sample
f.entry.instrument.detector  # raise if multiple
f.entry.instrument[NX_class.NXdetector]  # list/dict of detectors
f.entry[NX_class.NXmonitor]  # list/dict of monitors
```

I do not like the verbosity of using the enum `NX_class` which requires the prefix. We may consider directly using the classes as keys instead, removing the enum?

SimonHeybrock, 2022-09-08:
scipp/ess#139 is probably a good step forward. We may consider moving this into `scippneutron` if it turns out to be useful beyond ESS.

SimonHeybrock, 2022-10-24:
Not clear whether we should change `load_nexus` or just deprecate it, but with the new plotting approach (plopp), we want to avoid adding dummy TOF dimensions.

SimonHeybrock, 2023-04-17:
It seems like porting to `scippnexus.v2` would be a non-trivial effort, so I would suggest to move ahead with deprecating `load_nexus` as previously discussed, without refactoring for the future `scippnexus` API.

jl-wynen, 2023-07-11:
What do we do about data streaming? It currently uses `_load_nexus_json`.

SimonHeybrock, 2023-07-11:
Can we refactor to use `scippnexus`?

But also depends on how `Beamlime` will handle this.

jl-wynen, 2023-07-11:
> Can we refactor to use scippnexus?

We will have to do that at some point. But isn't this just as much effort as refactoring `load_nexus` which we concluded was not worth it?

Should we just disable streaming until Beamlime is at the point that it can use it? This would at least allow us to move forward.

SimonHeybrock, 2023-07-12:
> > Can we refactor to use scippnexus?
> 
> We will have to do that at some point. But isn't this just as much effort as refactoring `load_nexus` which we concluded was not worth it?

I meant to "refactor" to basically ScippNexus' recursive `__getitem__`, i.e., not keeping the current mechanisms that are in `load_nexus`, but ensuring that we can still run this with the JSON NeXus code.

SimonHeybrock, 2023-08-23:
Decision for removal has been made, deprecated as of release 23.08.

closed 2023-08-23 by SimonHeybrock