scipp/scippneutron #373: Saving reduced SANS data to NXCanSAS file

wpotrzebowski, 2022-09-30:
Current implementations of SANS data reduction workflows don't store reduced data to the file. 
I've explored some options and it seems that at minimum support to NXCanSAS file needs to be provided.

`sasdata` (part of SasView) provides https://github.com/SasView/sasdata/blob/master/sasdata/file_converter/nxcansas_writer.py, which can be used to save data from scipp (in the simplest case without any metadata bellow code can do the work)
```
from sasdata.file_converter.nxcansas_writer import NXcanSASWriter
from sasdata.dataloader.data_info import Data1D
writer = NXcanSASWriter()

data_1d = [Data1D(x=sc.midpoints(result.coords['Q']).values, y=result.values, dy=result.variances)]
writer.write(data_1d, 'test.h5')
```
However as discussed with @SimonHeybrock  most likely implementation will need become part of scipp anyway.

jl-wynen, 2022-09-30:
How widely is this format supported / used?

SimonHeybrock, 2022-09-30:
As far as I am aware it is a "standard" (it is even in the official NeXus docs: https://manual.nexusformat.org/classes/applications/NXcanSAS.html), i.e., I think scippneutron is a good place to support this.

SimonHeybrock, 2022-09-30:
As NXcanSAS is an "application definition" in NeXus, it may be worth considering if there is a more generic way to support these, as other techniques may want to use that as well.

SimonHeybrock, 2022-11-08:
I have just released scippnexus-0.4.0, which has experimental (and incomplete) support for writing NXcanSAS. Example:

```python
import scippnexus as snx
from scippnexus.definitions.nxcansas import NXcanSAS, SASentry, SASdata

with snx.File(test.nxs, 'w') as f:
    f['sasentry'] = SASentry(title='A test', run=12345)
    f['sasentry']['sasdata'] = SASdata(da, Q_variances='resolutions')
```

SimonHeybrock, 2022-11-25:
For now, initial requirements seem to be fulfilled (use scippnexux=0.4.2). Open a new issue when more features are required.

closed 2022-11-25 by SimonHeybrock