scipp/scippneutron #411: Transformations to `(Qx, Qy, Qz)` and `hkl`

SimonHeybrock, 2023-02-09:
- We have a transformation to `Q`, the magnitude of $\vec Q$. For single-crystal diffraction a transform to $\vec Q$ will be required.
- A similar case is conversion to `hkl`.This requires knowledge of the crystal orientation matrix. See for example https://docs.mantidproject.org/v6.1.0/concepts/Lattice.html.

jl-wynen, 2023-02-23:
Waiting for a release of Scipp with the inversions of transforms.

nvaytet, 2023-02-23:
To get the SANS `Qx`, `Qy`, would we just extract the components of the $Q$ vector?

jl-wynen, 2023-02-23:
The current implementation produces `Qx`, `Qy`, `Qz` separately. So you can just take those as individual coords. (Provided the maths is correct for sans(?))

nvaytet, 2023-02-23:
we might want to do something specific for SANS anyway, because they probably also want to take gravity into account.

SimonHeybrock, 2023-02-23:
I am not sure this is what you need for SANS. I think it is not even clear if the common SANS Qx/Qy formalism makes sense for LOKI without changes, since it has some quite large scattering angles. Should check with Wojciech.

closed 2023-03-24 by jl-wynen