scipp/scippneutron #458: Add a loader for mcstas files?

nvaytet, 2023-09-19:
A simple loader for ascii and hdf mcstas files was added in the [dmsc summer school repo](https://github.com/ess-dmsc-dram/dmsc-school/blob/main/4-reduction/load.py).

I could be useful to generalise this and make it into a loader for mcstas files.

One issue is that the files contain the `x` and `y` positions of the events, but not the `z` positions (columns are `["p", "x", "y", "n", "id", "t"]`).
Those, I guess, are deduced from where the detector is placed in the instrument.
So we would need to parse the instrument geometry to deduce them... which does not sound like a nice exercise.

In addition, the `x` and `y` coordinates are in the coordinate system of the detector, iirc. So this also means that if the detector is tilted with respect to the beamline axis (e.g. 45 degrees), then we also need the instrument geometry to convert those to real `x,y,z` in cartesian space.

Ideally for Scipp, at the end of the day, we need a list of `x,y,z` position in cartesian space, with absolute values (not relative to a detector panel).

Can this be achieved without parsing mcstas instrument files?

SimonHeybrock, 2024-01-31:
Parts of this was probably implemented in https://github.com/scipp/essnmx.

YooSunYoung, 2024-02-08:
Greg said he has already tried to make a general one so let's have a look at that too...!

jokasimr, 2025-11-12:
Is this issue fixed by https://github.com/mccode-dev/McStasToX?