scipp/scippneutron #482: Track chopper cutouts corresponding to subframes in `tof.chopper_cascade`?

SimonHeybrock, 2024-01-12:
It become apparent in #472 that this will likely be required, since not all chopper cutouts are in use at any given time. To compute the time-of-flight origin we need to be able to tell, given a frame or subframe defined by and computed with `tof.chopper_cascade`, which openings of a given chopper gave rise to that subframe.

It is not 100% clear if this is the correct approach. Maybe there is a better way of computing or setting the TOF origin?

jokasimr, 2025-01-20:
Should we close this issue as #589 is merged?

SimonHeybrock, 2025-01-20:
@nvaytet ?

nvaytet, 2025-01-22:
Frame unwrapping and WFM is moving to `essreduce`, and we are indeed moving away from using `chopper_cascade`, so yes, we can close this one.

closed 2025-01-22 by nvaytet