scipp/scippneutron #486: Bad instrument view button formatting

jl-wynen, 2024-01-25:
In the rendered docs, the 'toggle outline visibility' button of instrument views has a bad icon:
[image]
It alternates between an exclamation mark and question mark.

nvaytet, 2024-01-26:
I've also noticed this. This is a new thing, it used to work.
I don't know if it's some css issue (maybe something similiar to what is discussed [here](https://github.com/googlecolab/colabtools/issues/1302)?) or something else.

Note that the other icons are ok (home and camera). The icon for the outline is the `codepen` icon. It's from the brands section. Maybe something is up with the brands-specific icons?

nvaytet, 2024-01-26:
If the brand icons are the issue, we could consider changing things round a little:
- Use the [cube](https://fontawesome.com/v5/icons/cube?f=classic&s=solid) icon for the outline toggle
- Use the [layer-group](https://fontawesome.com/v5/icons/layer-group?f=classic&s=solid) icon for the spatial cutting tool
- And it would also be nice to harmonise the look/feel by having an icon for the RGB axes instead of a unicode character. Would something like a [ruler](https://fontawesome.com/v5/icons/ruler-combined?f=classic&s=solid) do? Or maybe [this dice](https://fontawesome.com/v5/icons/dice-d6?f=classic&s=solid) although probably too similar to the cube?

Or:
- [border icon](https://fontawesome.com/v5/icons/border-style?f=classic&s=solid) for the outline toggle and the [dice](https://fontawesome.com/v5/icons/dice-d6?f=classic&s=solid) for the axes?