scipp/scippneutron #489: Compute time at sample

SimonHeybrock, 2024-01-30:
This is required for event data filtering for fast sample environment parameters, e.g., in ESSsans.

Should this be added to the regular coord transform graphs that ScippNeutron provides?

jokasimr, 2024-01-30:
Is the time at the sample $t_s = L_1/v$?

SimonHeybrock, 2024-01-30:
Almost, this is about the *absolute* time at the sample, so $t_0 + t_s$. It is probably easiest to calculate backwards from the detector position (i.e., we do not require $L_1$).

jokasimr, 2024-01-30:
Okay I see, so we should compute the absolute time like $t_{abs,\ s} = t_0 + t_s = t_0 + t_d - L_2 / v$?

SimonHeybrock, 2024-01-30:
> Okay I see, so we should compute the absolute time like tabs, s=t0+ts=t0+td−L2/v?

Yes, and we might go via wavelength, since $v$ is generally not required, but wavelength is computed in many cases already.

jokasimr, 2024-01-30:
Where do we get $t_0$ (the time when the neutron passes the point where L=0)?

PR #491 (2024-02-06): feat: compute absolute time at sample
Fixes #489
closed 2024-02-06 by jokasimr