scipp/scippneutron #501: JSONGroup/JSONDataset not compatible with the latest json nexus template.

YooSunYoung, 2024-03-11:
It is not compatible with the latest json nexus templates.

For example,
https://github.com/scipp/scippneutron/blob/0988a279a8b2fc480bca867bdc4a06233d0e4e2b/tests/io/json_nexus_examples/entry.json#L1-L24

Now looks like:
```json
{
  "children": [{
      "name": "entry",
      "type": "group",
      "attributes": [{ "name": "NX_class", "dtype": "string", "values": "NXentry"}],
      "children": [{
          "module": "dataset",
          "config": { "name": "title", "values": "$TITLE$", "type": "string"}
        }]
    }]
}
```

We should make it compatible and update the examples.

SimonHeybrock, 2024-03-14:
I think this was clarified in our f2f, or is there still some unclear aspect?

jokasimr, 2024-03-14:
This comment was posted before we talked.

However, I'm still not sure what to do there. Maybe we can have a quick call today to figure it out.

jokasimr, 2024-03-18:
Should we continue to support the old json schema for specifying nexus structure, or should we only support the new schema?

SimonHeybrock, 2024-03-18:
Only the new.

PR #502 (2024-03-19): fix: json nexus loader
Fixes #501
closed 2024-03-19 by jokasimr