scipp/scippneutron #508: explore_data tutorial instrument_view with issue

jorobledo, 2024-04-29:
Hi all,

Was trying out `scippneutron` and following up with the tutorial [1_exploring-data.html](https://scipp.github.io/scippneutron/tutorials/1_exploring-data.html) and encountered two small issues which I worked around. I installed scippneutron by `pip install scippneutron[all]`  in a new pyenv on Ubuntu.

- When wanting to visualize instrument with `scn.instrument_view()` missing `pythreejs` dependency (installed it manually afterwards).
- Then got an error on the plotting:
[image]
which was solved by modifying `scene = fig.children[0].canvas.scene` to `scene = fig.canvas.scene` in `instrument_view.py`. 

Then it worked smoothly., but might as well report it.
Nice job btw.!

jl-wynen, 2024-04-29:
Concerning the first point, the `all` extra is a bit misleading. If you want interactive (3d) plots in Jupyter, you need the `interactive` extra: `pip install scippneutron[all, interactive]`. I'm going to change this in the docs.

nvaytet, 2024-04-29:
> The second issue was actually fixed in #505. We should make a release...

I was waiting for #507 to make a release. @MridulS what's the status there?

PR #509 (2024-04-30): Remove [interactive] pip install option
Fixes #508
closed 2024-04-30 by jl-wynen