scipp/scippneutron #531: Ensure two-theta calculation with gravity works if incoming beam is not perpendicular to gravity

SimonHeybrock, 2024-07-10:
Related to scipp/essreflectometry#14.

jl-wynen, 2024-07-10:
This may be easy:
1. Remove y-component of `incoming_beam` to define z-axis.
2. Use gravity to define the y-axis.
3. Compute angle between `incoming_beam` and x-z-plane.
4. Compute theta based on x-y-z-system and subtract above angle.

SimonHeybrock, 2024-07-10:
> Remove y-component of incoming_beam to define z-axis.

... assuming we never have a beam that is (approx) along y?

jl-wynen, 2024-07-10:
> > Remove y-component of incoming_beam to define z-axis.
> 
> ... assuming we never have a beam that is (approx) along y?

Yes. We could have a safety check to disallow beams below a certain angle from the y-axis.

Do you think this is an issue in practice? This issue is mostly about reflectometry. Given specular reflection and a beam coming down the y-axis, this would mean that the detector is also position along the y-axis, i.e., in the incoming beam.

SimonHeybrock, 2024-07-10:
> Do you think this is an issue in practice? This issue is mostly about reflectometry. Given specular reflection and a beam coming down the y-axis, this would mean that the detector is also position along the y-axis, i.e., in the incoming beam.

My thought was that someone might define Z as "up". Adding a simple check (say about the length of the incoming beam after removing Y?) should be sufficient?

jl-wynen, 2024-07-10:
Ok. Then you can project the incoming beam onto a plane perpendicular to gravity. Since we only care about the direction, not the length, this should work.

PR #544 (2024-09-03): Support non-orthogonal inputs when computing scattring angles with gravity
Fixes #531
closed 2024-09-03 by jl-wynen