scipp/scippneutron #533: Make pooch a soft dependency?

nvaytet, 2024-08-01:
When importing scippneutron without having installed `pooch`, we get the following
```
File [~/code/path/scippneutron/data/__init__.py:42](http://localhost:8889/home/nvaytet/code/path/scippneutron/data/__init__.py#line=41)
     15     import pooch
     17     return pooch.create(
     18         path=pooch.os_cache('scippneutron'),
     19         env='SCIPPNEUTRON_DATA_DIR',
   (...)
     38         },
     39     )
---> 42 _pooch = _make_pooch()
     45 def bigfake():
     46     return get_path('bigfake.nxs')

File [~/code/path/scippneutron/data/__init__.py:15](http://localhost:8889/home/nvaytet/code/path/scippneutron/data/__init__.py#line=14), in _make_pooch()
     14 def _make_pooch():
---> 15     import pooch
     17     return pooch.create(
     18         path=pooch.os_cache('scippneutron'),
     19         env='SCIPPNEUTRON_DATA_DIR',
   (...)
     38         },
     39     )

ModuleNotFoundError: No module named 'pooch'
```

I think we should make `pooch` a soft dependency, as one should be able to use `scippneutron` without it.

jl-wynen, 2024-08-05:
Pooch is a hard dependency: https://github.com/scipp/scippneutron/blob/a403f957148c9154f1f21aed8c303fa3fb99e795/pyproject.toml#L37

How did you install ScippNeutron that it didn't install pooch?

closed 2024-08-05 by jl-wynen

nvaytet, 2024-08-09:
> Pooch is a hard dependency:
> 
> https://github.com/scipp/scippneutron/blob/a403f957148c9154f1f21aed8c303fa3fb99e795/pyproject.toml#L37
> 
> How did you install ScippNeutron that it didn't install pooch?

I have a local clone of `scippneutron` and no `pooch` was installed in my conda environment.

nvaytet, 2024-08-09:
Do we have the same problem in all the technique specific packages? I'm quite sure that `esssans` also needs `pooch` on import?
e.g. https://github.com/scipp/esssans/blob/6e5fa3ac8712402014255e22ab0f38198192e5c2/src/ess/loki/__init__.py#L6

jl-wynen, 2024-08-09:
Yes, this is a problem everywhere. Simon started fixing it in ESSdiffraction. But we need to go through all packages and fix it.

SimonHeybrock, 2024-08-09:
Was fixed here: https://github.com/scipp/essdiffraction/pull/79

PR #534 (2024-08-05): Make pooch an optional dependency
Fixes #533

If accepted, I am going to move the new CI job to the template.