scipp/scippneutron #535: Update to use new Optional-handling in Sciline

jl-wynen, 2024-08-05:
After locking dependencies, I get errors like 
```python
___________ test_frame_period_is_pulse_period_if_not_pulse_skipping ____________

    def test_frame_period_is_pulse_period_if_not_pulse_skipping() -> None:
        pl = sl.Pipeline(unwrap.unwrap_providers())
        period = sc.scalar(123.0, unit='ms')
        pl[unwrap.PulsePeriod] = period
>       assert_identical(pl.compute(unwrap.FramePeriod), period)

tests/tof/unwrap_test.py:41: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
.tox/py310/lib/python3.10/site-packages/sciline/pipeline.py:90: in compute
    return self.get(tp, **kwargs).compute()
.tox/py310/lib/python3.10/site-packages/sciline/pipeline.py:138: in get
    graph = to_task_graph(self, targets=targets, handler=handler)
.tox/py310/lib/python3.10/site-packages/sciline/data_graph.py:217: in to_task_graph
    out[key] = handler.handle_unsatisfied_requirement(key)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <sciline.handler.HandleAsBuildTimeException object at 0x7f17d6fe8d30>
tp = typing.Optional[scippneutron.tof.unwrap.PulseStride], explanation = ()

    def handle_unsatisfied_requirement(self, tp: Key, *explanation: str) -> NoReturn:
        """Raise an exception when a type cannot be provided."""
>       raise UnsatisfiedRequirement('No provider found for type', tp, *explanation)
E       sciline.handler.UnsatisfiedRequirement: ('No provider found for type', typing.Optional[scippneutron.tof.unwrap.PulseStride])
```

So it seems that ScippNeutron is incompatible with the new handling of `Optional` in Sciline.

I don't know why this was not caught be nightly tests.They are all passing.

SimonHeybrock, 2024-08-05:
For some reason the nightly tests are using sciline==24.4.1, see https://github.com/scipp/scippneutron/actions/runs/10241440414/job/28329713317.

SimonHeybrock, 2024-08-05:
Sciline missing here? https://github.com/scipp/scippneutron/blob/a403f957148c9154f1f21aed8c303fa3fb99e795/.copier-answers.yml#L8

closed 2024-08-05 by jl-wynen