scipp/scippneutron #540: `scippneutron.io` no longer accessible directly

nvaytet, 2024-08-15:
[This change](https://github.com/scipp/scippneutron/commit/1b05f21b3058fd1857a62b618869f29be817cf06#diff-891780d06036ed4a7086c20e26743b96f75f3b4132e2a2a8ee4612b9f11b2566L43) appears to have removed `scippneutron.io` from the `__init__.py`, meaning that before we could do
```Py
import scippneutron as scn
scn.io.save_xye(...)
```
Now we have to do:
```Py
from scippneutron.io import save_xye
save_xye(...)
```

Should we make the `io` module accessible again?

SimonHeybrock, 2024-08-15:
I ran into the same. I fixed ESSdiffraction [here](https://github.com/scipp/essdiffraction/pull/79), but unless the module is expensive to import or has extra dependencies I guess we can add it back in?

jl-wynen, 2024-08-19:
Interesting. It looks like `io` was never explicitly exported (or imported). So 
> ```python
> import scippneutron as scn
> scn.io.save_xye(...)
> ```
only worked accidentally.

PR #555 (2024-09-19): Import and export io module
Fixes #540
closed 2024-09-19 by jl-wynen