scipp/scippneutron #597: Plots in the processing choppers notebook don't show all datat

jl-wynen, 2025-01-31:
The plateau plots in https://scipp.github.io/scippneutron/user-guide/chopper/processing-nexus-choppers.html don't show the data properly. This is because of https://github.com/scipp/plopp/issues/407

nvaytet, 2025-01-31:
In that notebook, can't we use
```Py
import plopp as pp

pp.plot({'a': a, 'b': b})
```
instead of going through MPL?

nvaytet, 2025-01-31:
Ot just use a DataGroup?

```Py
def plot_plateaus(raw_data: sc.DataArray, plateaus: sc.DataArray) -> None:
    to_plot = sc.DataGroup({'raw_data': raw_data})
    for plateau in plateaus:
        i = plateau.coords['plateau'].value
        to_plot[str(i)] = sc.DataArray(
            plateau.data.broadcast(dims=['time'], shape=[2]),
            coords={'time': plateau.coords['time']},
            name=f'Plateau {i}')
    return to_plot.plot()
```

jl-wynen, 2025-01-31:
We can. But that makes handling markers, colors, etc more awkward.

nvaytet, 2025-02-01:
We can either let `plopp` do that for us, or we can use https://scipp.github.io/plopp/user-guide/plot-types/line-plot.html#Customizing-line-styles-for-multiple-entries ?

PR #598 (2025-02-03): Fix plateau plots to use proper limits
Fixes #597
closed 2025-02-03 by jl-wynen