scipp/scippneutron #612: Consider adding atomic masses

jl-wynen, 2025-03-11:
For PDF reduction, we need the masses of atoms. Scippneutron provides a number of neutron-scattering-related parameters of atoms. We may want to add masses to this list. See, e.g., https://gitlab.com/ase/ase/-/blob/master/ase/data/__init__.py?ref_type=heads#L82

If we don't do this, people could use, e.g., [ASE](https://wiki.fysik.dtu.dk/ase/index.html) to get the masses. But this is a rather large package just for getting some numbers.

MridulS, 2025-03-12:
Yeah, I think we should package them directly from source https://www.degruyter.com/document/doi/10.1515/pac-2015-0305/html#j_pac-2015-0305_tab_001 instead of depending on another package.

MridulS, 2025-03-12:
Even better https://www.ciaaw.org/atomic-weights.htm I guess?

PR #613 (2025-03-24): Add atomic weights and masses
Fixes #612

The exact choice of which value to use is complicated. But for most applications, going with the standard atomic weight should be fine.
closed 2025-03-24 by jl-wynen